RobotDynOfficial/RBDDimmer #1: Strange behaviour with my LED lamp

igorantolic, 2019-01-27:
These are test results with ESP8266 160 Mhz (and also same using 80 Mhz)

**With LED bulb (IKEA RYET LED 400lm, 5W, LED1628G5, 220V)**
0-28% nothing
29% bulb is on/of with frequency about 2-4 Hz
30% about 80% light
35-99% = maximum light.

**With classic bulb - wolfram - (which you canot buy anymore in my country)**
0-40% = nothing
40-100% = linear dimming

So great for old bulbs, terrible for LED.
**Is that as expected by design?**

garrettdowd, 2019-02-06:
I bought your device - https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html

I am trying to use this library with dimmable LEDs. However the code is not functioning as expected and the LEDs are flickering.

What dimming method does this library use? Skip Pulse? Partial Pulse?

VSneg, 2019-02-08:
> For LED bulbs you need a special dimmer AND dimmable LED bulbs. (c) @JimmyBondi 
Dimmer use the Pulse Position Modulation using Zero-Cross signal, in easy way saying)
https://oscarliang.com/pwm-ppm-difference-conversion/

garrettdowd, 2019-02-08:
@VSneg so you are saying that this library does not support PPM? Could the library be modified to support PPM?

VSneg, 2019-02-28:
@garrettdowd, sorry, but not, maybe in the future we make another library only for LED lamps.

VSneg, 2019-02-28:
@garrettdowd sorry, but not, maybe in the future we make another library only for LED lamps.

garrettdowd, 2019-02-28:
My DOB LED does support trailing edge dimming. Could you provide a visual example of what this code is doing the the AC waveform? onTimerISR() is a bit confusing.

VSneg, 2019-02-28:
@garrettdowd https://www.youtube.com/watch?time_continue=36&v=dXSpunGnRd8

VSneg, 2019-02-28:
@garrettdowd LED lamps got their own power driver that's why classical AC dimming may not work or give unexpected results. If you got a dimmable led bulb, it's better to check vendor's instruction to get the dimming working properly.

garrettdowd, 2019-02-28:
From the video, it looks like this is leading edge dimming? Most dimmable LEDs use trailing edge dimming from what I understand.

ageurtse, 2019-05-18:
Well, i build a leading edge dimmer with led light's which is working great. 
But i used a attiny dedicated for dimming. 
When dimming with this library on a ESP8266 i als notice some flickering. 
i think this is due of code that is holding up the dim timmer. 

this is a know problem