RobotDynOfficial/RBDDimmer #2: Blinking

GmasteR, 2019-02-09:
Incandescent lamp blinks on your software on the NodeMCU v3 system :(

saichedelic, 2019-02-14:
same here, nodemcu v1.

VSneg, 2019-02-28:
@GmasteR, @saichedelic Please, can you give us more information about:
What lamp do you use?
When does it start blink?
What kind of blink?

saichedelic, 2019-02-28:
@VSneg 
I'm trying it on a 120W AC bulb and various platforms.
The new lib seems to have fixed it. Didn't get much time to completely check it today.
Will do more thorough testing when i get time tomorrow. thanks.

Marc-Calis, 2019-03-14:
Lamp Blinks (your latest software) on Wemos mini D1
Lamp 50 hz 220V
Start blink the moment you turn dimmer on
Blink very slow, every ± 15 seconds

tilman, 2019-03-29:
for me aswell.
Using an ESP32 dev board, with a 40W Edison Bulb at 50 hz 220V

blinking only occurs above 90% dimming value, but then about 1 to 3 times a second, but pretty irregularly with no specific pattern.

tilman, 2019-03-29:
Made a video of it (it's a zip because github does not support .m4v extension)
[IMG_5096.m4v.zip](https://github.com/RobotDynOfficial/RBDDimmer/files/3024591/IMG_5096.m4v.zip)
in the middle of the video i switched to 50% so flickering goes away

skokan8, 2020-07-13:
Did you solved this problem? Library works fine, but when i set more then 90% it is blinking

viewpointsa, 2022-09-28:
I have the same problem when value upper 90%, with arduino uno, RobotDyn module HL, in 50Hz 230V

fabianoriccardi, 2022-10-01:
@viewpointsa Had you tried [Dimmable Light](https://github.com/fabianoriccardi/dimmable-light) library? If you still had issues, you can try to adjust the interval which sets when the light is considered "fully on".