RobotDynOfficial/RBDDimmer #3: Zero Cross

AlpDmr, 2019-02-11:
I am trying to use your Module with this library for dimming 1400W AC fan but there are some frequency problems on lower voltages. It is probably about zero cross detection and to fix this problem I found zero-cross detection codes[ into your site](https://robotdyn.com/pub/media/0G-00005677==Mod-Dimmer-5A-1L/DOCS/Glow.ino) but while Dimming level (0-128)  0 = ON, 128 = OFF in that sketch, in official library we are using % values for dimming. So how to adapt [this](Dimming level (0-128)  0 = ON, 128 = OFF) codes to RBDDimmer library?

AlpDmr, 2019-02-12:
While I try to add these codes to my sketch with;
```
void setup(){
  attachInterrupt(digitalPinToInterrupt(2), zero_crossing, RISING); // I tried ports 2, 3, 18, 19, 20 and 21 as well
}
```
I am getting below error;
> WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':
> (.text+0x0): multiple definition of `__vector_5'
> libraries\RBDdimmer\RBDdimmer.cpp.o (symbol from plugin):(.text+0x0): first defined here
> c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions
> collect2.exe: error: ld returned 1 exit status
> exit status 1
> Error compiling for board Arduino/Genuino Mega or Mega 2560.

VSneg, 2019-02-28:
@AlpDmr  Sorry but this is really different codes, you couldn't adapt them