RobotDynOfficial/RBDDimmer #4: Mild pulsing at low power levels with AC Bulb

saichedelic, 2019-02-14:
Driving a 120W AC bulb with an arduino,
There is a noticeable pulsing of the ac bulb at lower power levels and continues all the way near 50.
I'm pretty sure its there at other power levels too, its jut not that noticeable.
Same behaviour was seen with the Mega which led me to guess its a trigger timing issue. (?)
Tried multiple bulbs and arduinos.

OjHond, 2019-02-19:
Also getting similar results. Have the RBDdimmer library running on Arduino Mega and I am experiencing flickering with both incandescent and dimmable LED downlighters. The lower the dim % the worse the flickering gets. I have tested the same module with the same bulbs with the Glow sketch and the 'for loop' runs through the dimming cycle without flickering. There seems to be something wrong with the timing of the triac. I am at 50Hz AC. Any help would be greatly appreciated.

saichedelic, 2019-02-20:
rbd's implementation in theory works with any frequency input, it would be the library to use. In my case the freq is not constant coming out of a home inverter. The dimming wasn't choppy at all as compared to other libs where freq is hard coded. Has the same mild pulsing issue at lower power levels tho.
Hopefully they fix it soon...
In the mean time I am using this, https://github.com/circuitar/Dimmer
Serves a similar purpose but does not work with inputs having slightly varying freq.

VSneg, 2019-02-28:
@saichedelic, @Oj-Hond, We made some changes, can you test, please?)

Oj-Hond, 2019-02-28:
@saichedelic  weird, i was getting better results with the freq hard coded.

@VSneg thanks, this is a huge improvement, still very minuscule flicker but not very noticeable.

Oj-Hond, 2019-02-28:
@VSneg the new library isn't working very well with an ESP8266-12E/F. I'm getting very bad flicker through out levels

Oj-Hond, 2019-02-28:
Seems to only be working on Mega, won't compile for Uno/Nano. Getting the error below:

C:\Users\acer\Documents\Arduino\libraries\RBDdimmer\src\avr\RBDmcuAVR.cpp: In member function 'void dimmerLamp::timer_init()':

C:\Users\acer\Documents\Arduino\libraries\RBDdimmer\src\avr\RBDmcuAVR.cpp:74:48: error: 'OCRxAH_VALUE' was not declared in this scope

  if (DIMMER_TIMER != 2) OCRxAH(DIMMER_TIMER) = OCRxAH_VALUE;

                                                ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

saichedelic, 2019-02-28:
> @saichedelic, @Oj-Hond, We made some changes, can you test, please?)

Thank you! This did solve the issues with respect to the mild flicker over constant ip frequency across the power levels with my 120W AC Bulb..
It has also improved a lot with the UPS output as well, will share a sample video. 
For now I tried only with one dimmer, will try with 2 or so tomorrow and revert.

saichedelic, 2019-02-28:
> Seems to only be working on Mega, won't compile for Uno/Nano. Getting the error below:
>
> C:\Users\acer\Documents\Arduino\libraries\RBDdimmer\src\avr\RBDmcuAVR.cpp: In member function 'void [...]
@Oj-Hond 
I didn't face any compilation issues. Did you try removing the entire lib from the folder? I tested it on a Uno personally [compiled on linux btw]. Will try the EPS8266 and the mega and revert.

Oj-Hond, 2019-02-28:
Yes i did, but re-downloaded the library and re-installed. Its working now thanks

saichedelic, 2019-02-28:
nice. I will do more tests tomorrow. If all goes well,, will close the issue.

edgsc, 2019-03-20:
I get flickering on original SerialMonitorDim.ino on a AC Bulb for 12%.
1% to 11% - ON, no apparent difference between them
12% flickering
13% to 22% is OFF
23% to 40% very very low light to low light
1-11% approx equal to 53-54%; 55% is lighter then 11%
95% highest light
96-99% flickering

Any ideas? Thanks!

edgsc, 2019-03-20:
"Glow.ino" works perfectly (from i=8 to 120; 5-7 blinks).
But teh same AC Bulb with "SerialMonitorDim.ino" has a strange behavior.
1-12% ON
11-13% blinks
14-22% OFF
23% very weak light, until ~54%
55% is about the same light as 1-12%
95-99% blinks

Any ideas?
I'm from Brazil (60Hz), should I change something?