RobotDynOfficial/RBDDimmer #5: Interrupt Problem with arduino uno

Stefan3424, 2019-03-01:
hello, 
i try to attach an interrupt on pin3 (pin 2 is used for zerocrossin)
but the compilers says:
WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':
(.text+0x0): multiple definition of `__vector_1'
libraries\RBDDimmer-master\avr\RBDmcuAVR.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Uno.

can someone help me to use the second interrupt on pin3 ?

VSneg, 2019-03-04:
@Stefan3424  Sorry but on uno haven't external interrupts on pin3

Stefan3424, 2019-03-16:
no i want to use pin3 (arduino uno) for a encoder via
 attachInterrupt(digitalPinToInterrupt(3), EncoderISR, LOW);