RobotDynOfficial/RBDDimmer #6: Severe pulsing when using multiple dimmers.

garrettdowd, 2019-03-14:
I am using this library with a Arduino Uno (AVR ATMEGA328P). I have come across two main problems.

1) The frequency of dimming drops (pulses are more noticeable) as more dimmers are used (single zero-cross with multiple output/"PWM" pins). Beyond 3 dimmers the timing is not consistent enough to use with dimmable LEDS. The pulses are clearly visible on incandescent bulbs.

2) The timing is unreliable. Adding dimmers seems to alter the timing of the dimming and thus the brightness. So a power level of 50% with 2 dimmers is visibly different than a power level of 50% with 6 dimmers.

I need to be able to use 5 dimmers for my project and at this point the library seems unusable for this case.

Is this somehow related the the interrupt capabilities of the arduino? Can a faster timer interrupt be set?

jkprow, 2019-11-11:
@garrettdowd I'm encountering this issue as well. Did you come up with a solution?

garrettdowd, 2019-11-11:
Yeah, I used a different library. https://github.com/circuitar/Dimmer

jkprow, 2019-11-11:
That library works great, thanks for the tip.

HFsi, 2019-11-11:
Does anyone tested it in ESP8266?

fabiuz7, 2019-11-11:
It seems that Circutar's library doesn't work on ESP8266, it's only AVR compatible (it depends on specific hardware timer). You may try [this alternative](https://github.com/fabiuz7/Dimmable-Light-Arduino)