RobotDynOfficial/RBDDimmer #10: Set power 0%

Goczilla, 2019-04-23:
Hello,

I am working with your hardware (AC Light Module Dimmer) and the RBDimmer library code, I control heating cartridge with your module. But, in normal mode, when I set the power to 0%, my cartridge is heating whereas it doesn't do that... Is it normal for you? For now, to make colder my cartridge, I have to set the power to 1% but it means I have always at less 1% of power sent to the cartbridge.

I thank you in advance for any help ;)

Mesmer88, 2019-04-24:
You can check my pull request. From what i saw in your case it won't enter the if loops in the library that change output pin state.

Goczilla, 2019-04-24:
I will try your solution and come back later to let you know if it is working. Thank you.

Goczilla, 2019-04-29:
If I add your modifications, I can't control anymore the card so I will add a relay to my circuit to make my 0%.

Mesmer88, 2019-04-29:
You mean it works fine but only 0% not working correctly?

Goczilla, 2019-04-29:
No, it seems for me my code doesn't work at all, I can't control the card so all the power is send to my cartridge. I don't have time to test again for now sorry. I just stay to the last updated library.

mrwnphs, 2021-10-17:
That's probably what I'm going to do also.  I already ordered the relay.

darrepac, 2022-04-02:
> Hello,
>
> I am working with your hardware (AC Light Module Dimmer) and the RBDimmer library code, I control heating cartridge with your module. But, in normal mode, when I set the power to 0%, my [...]

Yes the library is not well written: here a post I wrote with the corrections I made to solve this (and more)
https://blocnote360.wordpress.com/2022/03/20/routeur-solaire/ (sorry in french)
Hope it helps people in the same situation