RobotDynOfficial/RBDDimmer #14: Examples do not work with ESP8266

BlueBackbite, 2019-05-21:
Attempting to upload and run the SimpleToggleDimmer example on an ESP8266 throws an ISR not in IRAM! error.

I'm using Arduino IDE 1.8.9 with an Adafruit Feather HUZZAH 8266.

Cleared all the memory on the board and reflashed. 

I see ICACHE_RAM_ATTR is used in source. 
Any suggestions for troubleshooting from here?

`ISR not in IRAM!

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffc0 offset: 01b0
3fffff40:  40202aed 00000001 3ffef264 40202b0f  
3fffff50:  0001c200 0000001c 00000000 3ffee578  
3fffff60:  3fffdad0 3ffee520 00000005 401006c6  
3fffff70:  3fffdad0 00000001 3ffee520 4020123d  
3fffff80:  0001c200 0000001c 3ffee2b8 40201276  
3fffff90:  feefeffe 3ffee520 3ffee2b8 4020104a  
3fffffa0:  feefeffe 00000000 3ffee548 402019e0  
3fffffb0:  feefeffe feefeffe 3ffe8500 401009b1  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld`

BlueBackbite, 2019-05-22:
Out of frustration I switched to an ESP32 board and have no issues.
Unsure whether this is a problem with the ESP8266 being an single core board or not.

mkleijn, 2019-06-03:
Had the same thing. First resolved it by downgrading the ESP8266 board version to 2.5.0 in the board manager. However, the actual solution is adding in the file [src/esp8266/RBDmcuESP8266.cpp](https://github.com/RobotDynOfficial/RBDDimmer/blob/master/src/esp8266/RBDmcuESP8266.cpp#L139) the attribute `ICACHE_RAM_ATTR  `in front of the function `isr_ext()`, in the same way as it is in the function `onTimerISR()` (so `void ICACHE_RAM_ATTR isr_ext()`)

more background info on why you can find [here ](https://forum.arduino.cc/index.php?topic=616264.0)

HFsi, 2019-07-26:
I can't get it to work, no errors, but nothing happens.
I'm using NodeMCU 1.0, Arduino IDE 1.8.9, ESP8266 2.5.0 and the ICACHE_RAM_ATTR modification.
I can't even get the dimmer's led to turn on.
Any ideas?

Hoefnix, 2019-07-26:
I gave up, it is not a well supported product I guess. 

Went with this approach much cheaper and works perfectly to regulate the heat-lamp in a terrarium (https://www.instructables.com/id/230VAC-Dimmer-Controlled-by-Arduino-Raspberry-Pi/  )

theGeekyLad, 2020-12-27:
@Hoefnix can I avoid the photoresistor intermediary someway?

xmoulin, 2021-01-30:
Same, doesn't work with and ESP8266 Wemos D1 Mini. But work fine with ans [ESP32  Wemos D1 Mini](https://fr.aliexpress.com/item/4000650306925.html?spm=a2g0s.9042311.0.0.603f6c37OULACs)

anjana55, 2021-02-27:
This works on Node MCU 1.0. All you have to do is to add "ICACHE_RAM_ATTR" in between the "void  isr_ext()" RBDmcuESP8266.cpp file.

Like this:
void ICACHE_RAM_ATTR isr_ext()

xmoulin, 2021-02-28:
Thank you so much, it works perfectly! Thx @anjana55 , Thx @Github, Thx @Internet!

9ghost9, 2021-06-10:
> This works on Node MCU 1.0. All you have to do is to add "ICACHE_RAM_ATTR" in between the "void isr_ext()" RBDmcuESP8266.cpp file.
> 
> Like this:
> void ICACHE_RAM_ATTR isr_ext()

Thank you for this . Frustration relief :)

pritam2016bakshi, 2023-03-25:
> Thank you so much, it works perfectly! Thx @anjana55 , Thx @github, Thx @internet!

can you explain, how to setup you?

pritam2016bakshi, 2023-03-26:
> heth
How to