RobotDynOfficial/RBDDimmer #20: Power adjustment

robertowiest, 2019-07-23:
Hello, I am using your example https://github.com/RobotDynOfficial/RBDDimmer/blob/master/examples/SerialMonitorDim/SerialMonitorDim.ino with ESP32, very good. However, I have some doubts about using% between 0-100. By setting any percentage between 5-13 my lamp is flashing and between 14-22 it is off. Is that correct? Or am I implemented something wrong?

emrebilgin, 2021-03-03:
hi there.

i'm facing same issue.  is there a fix/workaround for this situation?

anjana55, 2021-03-03:
> Hello, I am using your example https://github.com/RobotDynOfficial/RBDDimmer/blob/master/examples/SerialMonitorDim/SerialMonitorDim.ino with ESP32, very good. However, I have some doubts about [...]

This maybe due to your hardware. See over here for a good explanation of how this works: https://forum.arduino.cc/index.php?topic=111196.0

mrwnphs, 2021-10-17:
I have a similar problem and I'm only powering an incandescent lamp.  The recommendation from anjana55 does not address this issue.