RobotDynOfficial/RBDDimmer #23: arduino nano compile error

carloperatoner, 2019-10-14:
I have an issue with an Arduino nano that I am using to control a laser cutter chiller.    The nano has two temperature probes, a flow sensor and a triac board for control of the cooling fan.   I am using D6 for the temperature probes (no interrupt needed) D2 for the zerocrossing input (interrupt 0) for the triac and D4 for the control line and D3 (interrupt 1) for the flow sensor.

This is the compile error I receive:
WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':
(.text+0x0): multiple definition of `__vector_1'
libraries\RBDDimmer-master\avr\RBDmcuAVR.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Nano.

It relates to this line in my code assigning the interrupt to for the flow sensor.

  attachInterrupt(1,Flow,RISING);

Full code is listed below:

#include <LiquidCrystal.h>      // LCD Library
#include <OneWire.h>            // Temp Sensor Library
#include <DallasTemperature.h>  // Temp Sensor Library
#include <RBDdimmer.h>          // Fan Control Library   

#define USE_SERIAL  Serial

// Setup temperature sensors
#define ONE_WIRE_BUS 6 // Connection for temperature probes pin 6
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);
double inputtemp;
double outputtemp;
double deltatemp;

// Setup LCD
LiquidCrystal lcd(12, 11, 10, 9, 8, 7);  // LCD Setup

// Setup Flow Sensor
int flowpin = 3; // Connection for flow sensor
double flowrate;
volatile int count;
double flowpulse = 2.25; //flow rate per pulse 2.25ml 

// Setup Fan Speed Control
#define fancontrolPin 4 
#define zerocross 2
dimmerLamp dimmer(fancontrolPin);

void setup(void)
{
  Serial.begin(9600); //Begin serial communication
  Serial.println("Arduino Digital Temperature // Serial Monitor Version"); // Startup message for Temperature
  sensors.begin();

  pinMode(flowpin, INPUT_PULLUP);  // Setup flow sensor configuration
  attachInterrupt(1,Flow,RISING);

  dimmer.begin(NORMAL_MODE, ON);  //Setup Fan control state
  dimmer.setPower(10);
}

void loop(void)
{ 
  // Flow Sensor Setup
  count = 0; // Set flow rate counter
  interrupts();
  
  // Temperature Sensor Routine
  sensors.requestTemperatures();  
  inputtemp = sensors.getTempCByIndex(0); 
  outputtemp = sensors.getTempCByIndex(1); 
  deltatemp = outputtemp - inputtemp;
  
  lcd.begin(16, 2);
  lcd.print(" In / Out  Flow");
  lcd.setCursor(0, 1);
  lcd.print(inputtemp,1); // Input Temperature to LCD
  Serial.println("Input Temp: ");
  Serial.println(inputtemp); // Input Temperature to Serial
  
 
  lcd.print("/");
  lcd.print(outputtemp,1); // Output Temperature to LCD
  Serial.println("Output Temp: ");
  Serial.println(outputtemp); // Output temperature to Serial

  noInterrupts();

  // Flow Sensor Routine
  flowrate = ((count * flowpulse *60) / 1000);  // Convert pulses to Liters/Minute

  lcd.print("   ");
  lcd.print(flowrate,1); // Output flowrate to LCD
  Serial.println("Flow Rate: ");
  Serial.println(flowrate); // Output flowrate to Serial

  if (deltatemp >= 10)
  {
     deltatemp = 10;
  }

  if (deltatemp <= 0)
  {
     deltatemp = 0;
  }
  dimmer.setPower(int(deltatemp * 10));
   delay(1000);   //Update value every 1 sec.
}

void Flow()
{
  // Count flow sensor pulses
  count++;
 }

Any suggestions?   It seems that the library attaches to both pins for interrupts. 

Thanks in advance.

WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':

(.text+0x0): multiple definition of `__vector_1'

libraries\RBDDimmer-master\avr\RBDmcuAVR.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Nano.

fabiuz7, 2019-10-29:
Hi carloperator,
May you try this [library](https://github.com/fabiuz7/Dimmable-Light-Arduino) on Arduino Nano? It is an alternative solution to control dimmers

carloperatoner, 2019-10-29:
Thank you.    I used the library and it worked.

Tutotrial, 2022-05-17:
can i set the lamp brightness if i use these library? thank you @fabianoriccardi

fabianoriccardi, 2022-05-18:
> can i set the lamp brightness if i use these library?

Yes of course!

Tutotrial, 2022-05-26:
In function `dimmerLamp::dimmerLamp(int)':
(.text+0x0): multiple definition of `__vector_5'
 

please help me about this problem. thanks

Tutotrial, 2022-05-26:
can i try this library on Arduino Mega?
thanks

Tutotrial, 2022-05-26:
@fabianoriccardi

fabianoriccardi, 2022-05-26:
May you tell me the version (or the commit) of the library and the target board?