RobotDynOfficial/RBDDimmer #32: Doesn't work for ESP8266

sumancvb, 2020-06-15:
I tried using this lib for NodeMCU. It works for few seconds after that it shows this error.
Dimmer Program is starting...
Set value
lampValue ->  10%
lampValue ->  50%
lampValue ->  40%

Exception (0):
epc1=0x40201d7c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdd0:  40106e04 3ffe9294 3ffee5f4 00000020  
3ffffde0:  ffffffff 00000020 00000022 00000030  
3ffffdf0:  00000017 0000001d 00000050 00000022  
3ffffe00:  3fffc200 40106dcc 3fffc258 4000050c  
3ffffe10:  40004378 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 00000000 80000000  
3ffffe30:  20000000 3fff0818 80000000 203fc080  
3ffffe40:  80000000 3fffc6fc 3ffe8e0c 3fff081c  
3ffffe50:  00000274 003fc080 60000600 00000030  
3ffffe60:  3fffc200 40106c3c 00000020 40106c84  
3ffffe70:  40106e68 20000000 3fffc230 4000050c  
3ffffe80:  40000f83 00000030 0000001b ffffffff  
3ffffe90:  402034a6 00000023 ffffffff ffffffff  
3ffffea0:  3ffe8e0c 4021af38 00000000 bfffffff  
3ffffeb0:  ffffffff 3fffc6fc 3ffe8e0c 3ffe91f4  
3ffffec0:  3ffe89a0 015c891a 60000600 00000030  
3ffffed0:  00000012 4021af38 3ffe8e0c 00000000  
3ffffee0:  ffffffff 3ffee5f4 3ffe8e0c 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffe89b0 00000030  
3fffff00:  00000000 400042db 00000064 60000600  
3fffff10:  40004b31 3fff079c 000002f4 003fc000  
3fffff20:  40101f6e 3ffe89a0 3ffe9140 40107188  
3fffff30:  4021aa05 3ffe9140 3ffe89a0 015c891a  
3fffff40:  3fff079c 00001000 4021ae9e 00000008  
3fffff50:  00000000 00000000 4021af4b 3ffe91f4  
3fffff60:  3ffe89a0 006cca0b 3ffe89a0 60000600  
3fffff70:  402034ad 3ffe91f4 3ffe89a0 015c5ff0  
3fffff80:  402034f2 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe89b8 3fffdad0 3ffee6cc 4020250b  
3fffffa0:  40000f49 00031343 3fffdab0 40000f49  
<<<stack<<<

sumancvb, 2020-06-15:
I also decoded dump
`Exception 0: Illegal instruction
Decoding 29 results
0x40201d7c: isr_ext() at C:\Users\edith\Documents\Arduino\libraries\RBDDimmer-master\src\esp8266/RBDmcuESP8266.cpp line 141
0x40106e04: interrupt_handler at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 119
0x40106c3c: timer1_isr_handler at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 33
0x40106dcc: interrupt_handler at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 113
0x40106c3c: timer1_isr_handler at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 33
0x40106e68: interrupt_handler at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 135
0x40106c3c: timer1_isr_handler at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 33
0x40106c84: timer1_isr_handler at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_timer.c line 40
0x40106dcc: interrupt_handler at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring_digital.c line 113
0x4021af38: pm_get_sleep_type at ?? line ?
0x40101f6e: spi_flash_read at ?? line ?
0x40107188: pvPortZalloc at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x4021aa05: pm_set_sleep_time at ?? line ?
0x4021ae9e: pm_get_sleep_type at ?? line ?
0x4021ce20: pp_noise_test at ?? line ?
0x4021af4b: pm_get_sleep_type at ?? line ?
0x402034ad: ets_timer_handler_isr at ?? line ?
0x402034f2: ets_timer_handler_isr at ?? line ?
0x4020250b: loop_task at C:\Users\edith\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56`

Need help

pille, 2021-07-06:
retry with the patch from #45. it worked for me.

sivar2311, 2021-10-02:
It looks like the rbdimmer library is still buggy.
Check the issues in the [RobotDyn repository](https://github.com/RobotDynOfficial/RBDDimmer/issues/32.).
A lot of people complain about this problem on the ESP8266.