RobotDynOfficial/RBDDimmer #38: Use with STM32 Oficial ST Core 1.9.0

AxelGUILLAUMET, 2020-11-26:
Dear friends,

I want to use the lib with a BluePill but with the STM32 ST oficial core.

[image]

And I get this error, so to correct it I added those:

[image]

So the compiler go more further but I have now this error...

[image]

Is someone could compile the lib with the oficial ST Core?

Regards.

muhammadnurimansyah, 2021-10-30:
The same Problem even with the latest core 2.x.x

muhammadnurimansyah, 2021-10-30:
also i am using f4 series where the directory source is not available