RobotDynOfficial/RBDDimmer #42: Hello Everyone can someone help me. How to use RBDDIMMER with Sinricpro

jamsyogendra, 2021-03-06:
Hello Everyone,,, 
 "So lets talk about my Project" 

Wifi Module: ESP8266 E-09 
Server using Sinric Pro; 

In my project, I am controlling, 2 Relays: ON/OFF 
And RobotDyn 1 Channel dimmer module for Fan Regulation. 

Here is Codes: 
```
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <SinricPro.h>
#include <SinricProSwitch.h>
#include <SinricProDimSwitch.h>
#include <RBDdimmer.h>

#define SSID          ""                                              // WiFi SSID
#define PASS          ""                                              // WiFi Password

#define APPKEY        ""                                              // SinricPro AppKey
#define APPSECRET     ""                                              // SinricPro AppSecret

#define RELAY1_ID     ""                                              // SinricPro deviceId for relay1
#define RELAY2_ID     ""                                              // SinricPro deviceId for relay2
#define DIMSWITCH_ID  ""                                              // SinricPro deviceId for dimmer

#define RELAY1_PIN  D0                                                // PIN relay1 is connected to
#define RELAY2_PIN  D1                                                // PIN relay2 is connected to
#define DIMMER_PIN  D5                                                // PIN dimmer is connected to
#define ZEROCROSS   D6                                                // for boards with CHANGEBLE input pins

dimmerLamp dimmer(DIMMER_PIN, ZEROCROSS);                             //initialase port for dimmer for ESP8266, ESP32, Arduino due boards

SinricProSwitch &relay1       = SinricPro[RELAY1_ID];                 // declare SinricPro device for relay1 and add the device to SinricProSDK
SinricProSwitch &relay2       = SinricPro[RELAY2_ID];                 // declare SinricPro device for relay2 and add the device to SinricProSDK
SinricProDimSwitch &dimSwitch = SinricPro[DIMSWITCH_ID];              // declare SinricPro device for dimmer and add the device to SinricProSDK                

int  dimLevel = 100;     // keeps the current dimming value (0..100) | initialized to 100 at start
bool pwrState = false;  // keeps the dimmer on/off state            | initialized to false (which means off) at start

// callback for onPowerState (relay1)
bool onPowerStateRelay1(const String& deviceId, bool &state) {
  digitalWrite(RELAY1_PIN, state);                                    // turn RELAY1_PIN on/off (depending on state parameter)
  Serial.printf("[Relay1]: turn %s\r\n", state ? "on" : "off");       // just some output on serial monitor
  return true;
}

// callback for onPowerState (relay2)
bool onPowerStateRelay2(const String& deviceId, bool &state) {
  digitalWrite(RELAY2_PIN, state);                                    // turn RELAY1_PIN on/off (depending on state parameter)
  Serial.printf("[Relay2]: turn %s\r\n", state ? "on" : "off");       // just some output on serial monitor
  return true;
}

// callback for onPowerState (dimmer)
bool onPowerStateDimSwitch(const String& deviceId, bool &state) {
  pwrState = state;                                                   // store the requested state in global variable (this state will be used in onPowerLevelDimSwitch )
  dimmer.setPower(pwrState? dimLevel : 0);
  Serial.printf("[DimSwitch]: turn %s (level: %d)\r\n",               // just some output on serial monitor
                state ? "on" : "off", dimLevel);
  return true;
}

// callback for onPowerLevel (dimmer)
bool onPowerLevelDimSwitch(const String &deviceId, int &level) {
  dimLevel = level;                                                   // store requested level in global variable (this state will be used in onPowerStateDim )
  dimmer.setPower(level);
  Serial.printf("[DimSwitch]: set level %d (state: %s)\r\n",          // just some output on serial monitor
                level, pwrState ? "on" : "off");
  return true;
}

// setup part for pin initialization (just to keep setup() nice and clean)
void setupPins() {
  pinMode(RELAY1_PIN, OUTPUT);                                        // initialize RELAY1_PIN as OUTPUT
  pinMode(RELAY2_PIN, OUTPUT);                                        // initialize RELAY2_PIN as OUTPUT
  dimmer.begin(NORMAL_MODE, ON);                                      //dimmer initialisation: name.begin(MODE, STATE)
}

// setup part for connecting to WiFi (just to keep setup() nice and clean)
void setupWiFi() {
  Serial.printf("[WiFi]: Connecting to %s", SSID);                    // print to which WiFi we are trying to connect to
  WiFi.begin(SSID, PASS);                                             // start connecting to WiFi
  while (WiFi.status() != WL_CONNECTED) {                             // wait until WiFi has been connected
    Serial.printf(".");                                               //    meanwhile print a dot
    delay(250);                                                       //    every 250ms  
  }
  Serial.printf("connected\r\n");                                     // print to serial that we are connected to wifi now
}

// setup part for SinricProSDK (just to keep setup() nice and clean)
void setupSinricPro() {
  relay1.onPowerState(onPowerStateRelay1);                            // assign onPowerStateRelay1 callback function to relay1 device
  relay2.onPowerState(onPowerStateRelay2);                            // assign onPowerStateRelay2 callback function to relay2 device
  dimSwitch.onPowerState(onPowerStateDimSwitch);                      // assign onPowerStateDimSwitch callback function to dimSwitch device
  dimSwitch.onPowerLevel(onPowerLevelDimSwitch);                      // assign onPowerLevelDimSwitch callback function to dimSwitch device

  SinricPro.onConnected([]() {                                        // instead of writing a full callback function we use a lambda function which get called when ESP is connected to SinricPro
    Serial.printf("[SinricPro]: Connected\r\n");                      // and this function does: output to serial "[SinricPro]: Connected"
  });

  SinricPro.onDisconnected([]() {                                     // instead of writing a full callback function we use a lambda function which get called when ESP is disconnected from SinricPro
    Serial.printf("[SinricPro]: Disconnected\r\n");                   // and this function does: output to serial "[SinricPro]: Disconnected"
  });

  SinricPro.begin(APPKEY, APPSECRET, "testws.sinric.pro");                                 // initialize SinricProSDK with APPKEY and APPSECRET
}

void setup() {
  Serial.begin(115200);                                               // initialize Serial to baudrate 115200
  Serial.println();                                                   // print a single line (to get rid of the junk)
  setupPins();                                                        // call setupPin function
  setupWiFi();                                                        // call setupWiFi function
  setupSinricPro();                                                   // call setupSinricPro function
}

void loop() {
  SinricPro.handle();                                                 // do the magic...
}

```
Code compiling with zero error no any issues in compelling: 
the only problem is the codes,the not running in my ESP8266 

My serial monitor details are given below here: 

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

ISR not in IRAM!

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffec0 end: 3fffffc0 offset: 0000
3ffffec0:  feefeffe feefeffe feefeffe feefeffe  
3ffffed0:  000000fe 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 00000000 00ff0000  
3ffffef0:  5ffffe00 5ffffe00 feefeffe 00000000  
3fffff00:  00000002 00000005 00000005 4020e36e  
3fffff10:  40100816 feefeffe feefeffe 4020e380  
3fffff20:  4020f31d 3fff0e54 00000005 4020e889  
3fffff30:  00000000 00000000 535b000a 4020ef99  
3fffff40:  3ffe89b1 4020f501 ffffffff 3ffef4e4  
3fffff50:  3fffdad0 00000000 00000005 4020e938  
3fffff60:  4020c5f8 3ffef38c 3ffe89af 4020c471  
3fffff70:  4020c5f8 3ffef38c 3ffeeeb0 4020c4aa  
3fffff80:  3fffdad0 00000000 3ffef38c 402020b8  
3fffff90:  3fffdad0 00000000 3ffef38c 40202c71  
3fffffa0:  feefeffe feefeffe 3ffef4a4 4020df20  
3fffffb0:  feefeffe feefeffe 3ffe8500 40101111  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

ISR not in IRAM!

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

Can someone tell me about the Exception Decoder...

petanque, 2021-03-06:
[https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html#exception-decoder](url)

jamsyogendra, 2021-10-02:
```
Decoding stack results
0x401007de: puts(char const*) at C:\Users\jamsa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\libc_replacements.cpp line 117
0x4020aeac: __assert_func(char const*, int, char const*, char const*) at C:\Users\jamsa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_postmortem.cpp line 275
0x4020b3b9: __attachInterruptFunctionalArg(uint8_t, voidFuncPtrArg, void*, int, bool) at C:\Users\jamsa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 201
0x40209c88: String::~String() at C:\Users\jamsa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\WString.cpp line 125
0x4020b468: __attachInterrupt(uint8_t, voidFuncPtr, int) at C:\Users\jamsa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_wiring_digital.cpp line 236
0x402091d1: dimmerLamp::ext_int_init() at C:\Users\jamsa\OneDrive\Documents\Arduino\libraries\RBDDimmer-master\src\esp8266\RBDmcuESP8266.cpp line 62
0x4020920a: dimmerLamp::begin(DIMMER_MODE_typedef, ON_OFF_typedef) at C:\Users\jamsa\OneDrive\Documents\Arduino\libraries\RBDDimmer-master\src\esp8266\RBDmcuESP8266.cpp line 71
0x40202698: setup() at C:\Users\jamsa\OneDrive\Desktop\project sinric pro\sinricpro2relay - Copy\sinricpro2relay/sinricpro2relay.ino line 95
0x4020aa4c: loop_wrapper() at C:\Users\jamsa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 194

```

stephen-j-oleary, 2022-11-09:
@jamsyogendra Any luck solving this? I'm running into the same issue