RobotDynOfficial/RBDDimmer #44: 0% light flicker

KevinJordil, 2021-06-08:
Hello,

I use this library to turn on a lamp and when I set the percentage to 0%, it happens that the light turns on and then turns off directly after.

An idea where this problem can come from

9ghost9, 2021-06-10:
If you try 50% is the light on stable ?
How about -1% , is it off stable ?
1% -> 10% seems to low of power to turn on a lamp . The value scale in the library code is 1 -> 99 .

I am just experimenting with a 200W led flood lamp and I am getting a small buzzing with lots of flicker .
The only semi stable values seem to be between 70 - 88 % .

KevinJordil, 2021-06-10:
I haven't tried 50% in the long term.
By the way, I would like the lamp to stay off at night, but to turn on from time to time.

I'm not sure if putting -1 changes anything because in the function if it's values less than 1, the if is set to 1.

Maybe by modifying the code and letting a value of 0 pass it works?

9ghost9, 2021-06-11:
What hardware and method of control are you using ?
When you say turn on and off for time to time ,are you thinking pir/motion sensor ?

I am using the Wemos pro mini and a simple async Web server .
While testing I have found that -1 absolutely turns off the Flood leds.

bogomips, 2021-12-06:
With a Wemos board connected to a 2CH dimmer any value < 6 makes bulbs flickering and the light intensity is quite high.
if I set 6 lamps are off

KevinJordil, 2022-01-31:
@bogomips
I tried with 6 but my lamp lights up. However, with 1, it's really better than with 0. I'm going to test it over a longer period of time to see if the problem is solved

Grufoony, 2023-09-04:
I have the same problem, using 2 as minimum... have you solved it?

closed 2025-04-17 by KevinJordil