RobotDynOfficial/RBDDimmer #45: Please Fix the ESP portion of this Lib .

9ghost9, 2021-06-15:
I have added Change to the ISR ability and corrected what others have already suggested .

It is frustrating to purchase RobotDyn products only to have the firmware not function .

Here is the fixed Lib completed that will remove ESP flashing frustration  and also enable reasonable AC LED control .

https://drive.google.com/file/d/13wZMFQneHm8znYDxutUyMGcZuyUESraV/view?usp=sharing

RBDmcuESP8266.cpp

line: 9  int rise_fall_change = true;    // added change
line: 61 attachInterrupt(inPin, isr_ext,CHANGE);  // set to CHANGE  and works with incans or leds.

Note; From my testing and experience with the Robotdyn AC module , by adding a parallel load to Led fixtures  the top end percentage is not clipped at 80% but works correctly through to 100%

Thank you

canDry, 2021-07-25:
> It is frustrating to purchase RobotDyn products  
  
**This.**  So damn frustrating.  
  
**THANK YOU** ... got it working on ESP8266 with **YOUR** updated library!

9ghost9, 2021-07-26:
> > It is frustrating to purchase RobotDyn products
> 
> **This.** So damn frustrating.
> 
> **THANK YOU** ... got it working on ESP8266 with **YOUR** updated library!

hi 
Here is much better code , you will need the hw_timer.lib .

`// Import required libraries
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <hw_timer.h>

// network credentials
const char* ssid = "AP";  // ACCESS POINT
const char* password = ""; //ENTER PASSWORD IF DESIRED

const byte zcPin = D1;
const byte pwmPin = D6;

byte fade = 1;
byte state = 1;
byte tarBrightness = 255;
byte curBrightness = 0;
byte zcState = 0; // 0 = ready; 1 = processing;
void ICACHE_RAM_ATTR zcDetectISR ();
void ICACHE_RAM_ATTR dimTimerISR ();

String sliderValue = "0";

const char* PARAM_INPUT = "value";

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

const char index_html[] PROGMEM = R"rawliteral(
<!DOCTYPE HTML><html>
<head>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <title>ESP Web Server</title>
  <style>
    html {font-family: Arial; display: inline-block; text-align: center;}
    h2 {font-size: 2.3rem;}
    p {font-size: 1.9rem;}
    body {max-width: 400px; margin:0px auto; padding-bottom: 25px;}
    .slider { -webkit-appearance: none; margin: 14px; width: 360px; height: 25px; background: #FFD65C;
      outline: none; -webkit-transition: .2s; transition: opacity .2s;}
    .slider::-webkit-slider-thumb {-webkit-appearance: none; appearance: none; width: 35px; height: 35px; background: #003249; cursor: pointer;}
    .slider::-moz-range-thumb { width: 35px; height: 35px; background: #003249; cursor: pointer; } 
  </style>
</head>
<body>
  <h2>ESP Web Server</h2>
  <p><span id="textSliderValue">%SLIDERVALUE%</span></p>
  <p><input type="range" onchange="updateSliderPWM(this)" id="pwmSlider" min="0" max="255" value="%SLIDERVALUE%" step="2" class="slider"></p>
<script>
function updateSliderPWM(element) {
  var sliderValue = document.getElementById("pwmSlider").value;
  document.getElementById("textSliderValue").innerHTML = sliderValue;
  console.log(sliderValue);
  var xhr = new XMLHttpRequest();
  xhr.open("GET", "/slider?value="+sliderValue, true);
  xhr.send();
}
</script>
</body>
</html>
)rawliteral";

// Replaces placeholder with button section in your web page
String processor(const String& var){
  //Serial.println(var);
  if (var == "SLIDERVALUE"){
    return sliderValue;
  }
  return String();
}

void setup(){
  // Serial port for debugging purposes
  Serial.begin(115200);

  WiFi.softAP(ssid, password);
  IPAddress IP = WiFi.softAPIP();//

  pinMode(zcPin, INPUT_PULLUP);
  pinMode(pwmPin, OUTPUT);
  attachInterrupt(zcPin, zcDetectISR, RISING); // Attach an Interupt to Pin  (interupt 0) for Zero Cross Detection
  hw_timer_init(NMI_SOURCE, 0);
  hw_timer_set_func(dimTimerISR);

  // Route for root / web page
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/html", index_html, processor);
  });

  // Send a GET request to <ESP_IP>/slider?value=<inputMessage>
  server.on("/slider", HTTP_GET, [] (AsyncWebServerRequest *request) {
    String inputMessage;
    // GET input1 value on <ESP_IP>/slider?value=<inputMessage>
    if (request->hasParam(PARAM_INPUT)) {
      inputMessage = request->getParam(PARAM_INPUT)->value();
      sliderValue = inputMessage;

    }
    else {
      inputMessage = "No message sent";
    }
    Serial.println(inputMessage);
    request->send(200, "text/plain", "OK");
  });
  
  // Start server
  server.begin();
}
  
void loop() {
  int val; 
  val = sliderValue.toInt();
  if (val > 0){
   tarBrightness = val;
  }
}

void dimTimerISR() {
if (fade == 1) {
if (curBrightness > tarBrightness || (state == 0 && curBrightness > 0)) {
--curBrightness ;
}
else if (curBrightness < tarBrightness && state == 1 && curBrightness < 255) {
++curBrightness;
}
}
else {
if (state == 1) {
curBrightness = tarBrightness;
}
else {
curBrightness = 0;
}
}

if (curBrightness == 0) {
  state = 0;
  digitalWrite(pwmPin, 0);
}
else if (curBrightness == 255) {
  state = 1;
  digitalWrite(pwmPin, 1);
}
else {
  digitalWrite(pwmPin, 1);
}

zcState = 0;
}

void zcDetectISR() {
if (zcState == 0) {
zcState = 1;

if (curBrightness < 255 && curBrightness > 0) {
  digitalWrite(pwmPin, 0);
  
  int dimDelay = 29 * (255 - curBrightness) + 400;//400
  hw_timer_arm(dimDelay);
}
}
}`

canDry, 2021-07-26:
> > > It is frustrating to purchase RobotDyn products
> > 
> > 
> > **This.** So damn frustrating.
> > **THANK YOU** ... got it working on ESP8266 with **YOUR** updated library!
> 
> hi
> Here is much better code , you will need the hw_timer.lib .
> Find it here-> [https://github.com/Theb-1/ESP8266-wifi-light-dimmer](url)
> 

**THANK YOU** again!  Works great on an esp8266, and _doesn't need_ RobotDyn's library.   😁
  
Any chance you have one that works for ESP32?  This (esp8266) version requires "ets_sys.h which isn't found for ESP32.  I found an ESP32 timer library (hal_timer e.g. https://desire.giesecke.tk/index.php/2018/04/22/using-the-hw-timers-of-the-esp32/) but haven't had luck in using it for dimmer timing.

9ghost9, 2021-07-27:
I had not considered AC dimming with the Esp32 but I will take a look at it when I have some free time. 
I have found the Wemos mini pro with external to be excellent for this task combined with the RobotDyn Wemos shield.
RobotDyn customer service is very poor in regards to their shield and is why I dumped their .lib for the hw_timer.

9ghost9, 2021-09-15:
> **THANK YOU** again! Works great on an esp8266, and _doesn't need_ RobotDyn's library.
>
> Any chance you have one that works for ESP32? This (esp8266) version requires "ets_sys.h which isn't found [...]

So ,some good news , I went ahead and built an AC dimmer that is controllable with any Arduino board or ESP module . . It does not use isr's but rather  uses analogWrite and in the case of ESP32 LedcWrite for direct PWM control .

A very simple circuit to build as DIY !
Will probably need to go via PM from here if you need any more advice .

unaipuelles, 2022-03-08:
Hi,

Anyone knows if can be used with the board of Arduino Uno Wifi Rev2 that has the chip Atmega4809. I try to compile the library for this board and I can't. I also searched for other dimming modules and I can't find anything for this board.

Thank you!

PD: This message has nothing in common with this issue but I don't know where to ask this because the maintainers of this repo don't respond anymore.

Alan-Somebody, 2022-05-02:
> So ,some good news , I went ahead and built an AC dimmer that is controllable with any Arduino board or ESP module . . It does not use isr's but rather uses analogWrite and in the case of ESP32 [...]

@9ghost9 , sadly I can't see a way to PM you in here.  were you able to get this to work on an ESP32 by any chance?  I just bought their 16A/600V dimmer and very disappointed that I can't get it to work on my 'ESP32 Dev C' unit & that RobotDyn don't offer any working software or help!

Alan-Somebody, 2022-05-02:
Thank you very much @9ghost9 for replying and for the link.  I will take a look.

9ghost9, 2022-05-02:
There are a few cautions with this circuit .
The load line is constant /always live  and  L & N need to be wired correctly or short circuit will occur .

9ghost9, 2022-05-04:
> hi there, is there any progress with making this library working on ESP32? thx

Have you tried this link ?
https://github.com/RobotDynOfficial/Lib-RBD-Dimmer-for-ESP32/tree/master/RBDdimmerESP32

Alan-Somebody, 2022-05-05:
Hi @9ghost9 .  I did try the link yes.  The code compiles when I tell the Arduino IDE that I am using an Arduino Uno, but the code fails when I tell it I am using the ESP32.

9ghost9, 2022-05-05:
You need to install the RBDdimmerESP32 lib and install in your Arduino/library.

https://github.com/RobotDynOfficial/Lib-RBD-Dimmer-for-ESP32

I just tried and it compiles fine for Esp32 .

Sketch uses 208513 bytes (15%) of program storage space. Maximum is 1310720 bytes

coolstuff99, 2023-05-22:
> 

It compiles but @9ghost9 did you get it working?
I mean, I know code works because added a println in the main and returns from the serial monitor but no way to get bulb on, neither dimmable :(

9ghost9, 2023-05-23:
I gave up on RBD for the ESP modules .
I built the circuit from this demonstration video and am able to control AC & DC elements/fixtures with PWM .

This has perfect dimming from 0 -> 255

[https://www.youtube.com/watch?v=XdCvJ1wZ0bA](url)