RobotDynOfficial/RBDDimmer #51: ESP32 : code source works ?

davidmarli, 2022-01-30:
Hi, I used tu use this robotdyn dimmer with an esp8266 (wemos)
Yet, I change for a ttgo-t-display (esp32)
In RBDmcuESP32.cpp, I see line 6 was corrected : value true (same as in esp8266.h)
But line 61 or so : 

void dimmerLamp::ext_int_init(void) 
{
	int inPin = dimZCPin[this->current_num];
	pinMode(inPin, INPUT_PULLUP);
    attachInterrupt(inPin, isr_ext, RISING);  
}

Do I have to change for Falling or CHANGE ????

Last question : ALLdimmers is define at 50 or 30 (depends if it's in RBDXX.h or .cpp (value of 20 for 8266) Do I hve to change this value and by which one ? And someone can explain me what's the purpose of this value ?

I ask those questions, because I use a load of 2200W, and if I set the dimmer to 50%, the power raises to 1700W !!! not normal. it should be 1100W, or I don't understand the purpose of the dimmer.

Thank you.

davidmarli, 2022-11-02:
anyone ???

fabianoriccardi, 2022-11-02:
I think the main issue is that this library controls the dimmer by "time", but on AC the power doesn't scale linearly with time.

[Here](https://github.com/fabianoriccardi/dimmable-light) there is my dimmer library that provides such linearization. If you have any feedback or question let me know.

davidmarli, 2022-11-03:
Ok, I try to use your librairy.
How do I declare it in Visual studio code ?
Where do I copy all the sources files ?

Thank you.

Arulamir, 2024-08-10:
> I think the main issue is that this library controls the dimmer by "time", but on AC the power doesn't scale linearly with time.
> 
> [Here](https://github.com/fabianoriccardi/dimmable-light) there is my dimmer library that provides such linearization. If you have any feedback or question let me know.

hi, i use ur library but it still cannot bi used in my esp32. i use a newer esp32 board, so maybe its not compatible with ur library anymore