RobotDynOfficial/RBDDimmer #52: Not working on Arduino Nano 33 boards

ipener, 2022-02-08:
The source code unfortunately doesn't work with any Arduino Nano 33 board (IoT, BLE, BLE Sense). In case they aren't supposed to be supported, I'd suggest to update the `README.md` accordingly. [Here's a minimal implementation](https://github.com/ipener/arduino-nano-33-dimmer) that runs on all of these boards, tested with a [RobotDyn® 4-channel AC dimmer](https://robotdyn.com/ac-light-dimmer-module-4-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html).

rockydog1984, 2022-02-15:
Thanks @ipener 

I am having issues getting Arduino MKR 1010. It appears to be SAMD21 based so should work, but when I run example sketch "SerialMonitorDim" nothing prints so I assume its not running due to incompatibility.

Do you think this library should work on this device? How did you fix for Nano 33?

ipener, 2022-02-15:
It was faster to write a minimal working example than fixing it. If you want you can't try out the implementation from my repo but please note that I only tested it on the Nano 33 boards. If for instance the MKR 1010's processor is running on a different frequency than the Nano 33 IoT then you'd have to adapt the clock frequency.

rockydog1984, 2022-02-15:
Ah I see, thanks! I shall run it and see what happens :)

jonesek, 2023-01-01:
> It was faster to write a minimal working example than fixing it. If you want you can't try out the implementation from my repo but please note that I only tested it on the Nano 33 boards. If for [...]

Hi, Works wery well with IoT33. Thank you a lot . Well done.. 
Pavel