RobotDynOfficial/RBDDimmer #53: Timing issue

darrepac, 2022-03-05:
Hi 

Looking at the code for ESP8266, I found something strange about ISR timer:
`void dimmerLamp::timer_init(void)
{
	timer1_attachInterrupt(onTimerISR);
	timer1_enable(TIM_DIV16, TIM_EDGE, TIM_SINGLE);
	timer1_write(timeoutPin); //100 us
}`
The 100us as comment sounds to be the right value (indeed with 50Hz signal, we have 10ms demi-period and so with 100 steps (or %) allowed, we need a 10 000/100 = 100us timer)

yet above is written:
`int timeoutPin = 435; // 80us`

In fact 435 means (0.2*435) = 87us... but why? it should be 100us and so 500... I don't understand it and moreover there is a difference in the code itself: at one place it is written 100us, on another 80us...

Timer being shorter that it should (in my opinion!), it could explain some flickering or at least some not 0W consumption at dimmer = 0

@VSneg as you seem to be one of the writer of this library, I would appreciate your feedback

darrepac, 2022-04-02:
Hello

A blog post about how I debugged and corrected the library here : https://blocnote360.wordpress.com/2022/03/20/routeur-solaire/ (sorry, in french...)

yokusan, 2022-07-12:
> Hello
> 
> A blog post about how I debugged and corrected the library here : https://blocnote360.wordpress.com/2022/03/20/routeur-solaire/ (sorry, in french...)

Dear Darrepac, you should try to integrate your corrections into this original Robotdyn Library. All users of the library would benefit from your modifications.

yokusan, 2022-07-12:
@VSneg please integrate the modifications by Darrepac into your Dimmer library.

davidmarli, 2022-08-24:
Hi, and for esp32 version ?

fbndny, 2024-05-19:
> Hello
> 
> A blog post about how I debugged and corrected the library here : https://blocnote360.wordpress.com/2022/03/20/routeur-solaire/ (sorry, in french...)

Thanks a lot!