RobotDynOfficial/RBDDimmer #54: RBDDimmer and MAX6675, Reboot itself

microitech, 2022-03-28:
Hi, everybody, 
When I use just one function to set dimmer power everything works fine, but when I add another function on the loop like read temperature with Max6675 then Arduino seems to reinitialize every 10 seconds.

That's my code:

//Libraries
#include <RBDdimmer.h>//https://github.com/RobotDynOfficial/RBDDimmer
#include "max6675.h" 
//Parameters
const int zeroCrossPin  = 2;
const int acdPin  = 3;
int power  = 0;
dimmerLamp acd(acdPin);

//Temp sensor 
int SO = 12;
int CS = 11;
int sck = 13;
MAX6675 module(sck, CS, SO);
float temperature = 25;

void setup() {
  //Init Serial USB
Serial.begin(9600);
Serial.println(F("Initialize System"));
acd.begin(NORMAL_MODE, ON);

}

void loop() {
  // put your main code here, to run repeatedly:
  //temp();
  dimming();
  

  
  }
    void temp() {
temperature = module.readCelsius(); 
  
   

  }

  void dimming(){
if (temperature <= 30) {
    power = 20;
    acd.setPower(power); // setPower(0-100%);
     Serial.print("lampValue -> ");
     Serial.print(acd.getPower());
     Serial.println("%");
     
    

  }
  if (temperature > 30) {
    power = 0;
    acd.setPower(power); // setPower(0-100%);
     Serial.print("lampValue -> ");
     Serial.print(acd.getPower());
     Serial.println("%");
     

  }
   Serial.print("Temperature: ");
 Serial.println(temperature);
   delay(500);
  }

yokusan, 2022-03-29:
What happens if you add a delay:

microitech, 2022-04-04:
> What happens if you add a delay after the temp measurement: void temp() { temperature = module.readCelsius(); delay(200);
> 
> and you will have to remove the // before the temp() in the loop

I have add delay and it's still the same. 

I had comment temp() just to test if was working and then i forgot to delete before posting here.

horemansp, 2022-04-24:
Did you found a solution for this? I'm running into the same issue. 
It says: "ISR not in IRAM!" and keeps rebooting.
I'm using a Wemos D1 mini pro.

horemansp, 2022-04-24:
I made the following changes to the code and now It does not reset anymore. (with the latest esp8266 library 3.0.2)
device: Wemos D1 mini pro

esp8266/RBDmcuESP8266.h
line 25
void ICACHE_RAM_ATTR isr_ext(void);

eps8266/RBDmcuESP8266.cpp
line 139
void ICACHE_RAM_ATTR isr_ext()

baldevg, 2023-12-13:
> Did you found a solution for this? I'm running into the same issue. It says: "ISR not in IRAM!" and keeps rebooting. I'm using a Wemos D1 mini pro.
> 
> I downgraded my esp8266 boards library to v2.4.2, then I don't have a problem. Obviously that's not the desired solution...

Thank you very much dear..

baldevg, 2023-12-13:
Thank you very much dear