RobotDynOfficial/RBDDimmer #62: Library Interrupts Conflict

siteswapjuggler, 2022-12-02:
As far as I understand this library only use interrupt on pin 2, but the declaration of it is totally blocking other interrupt use tested on Uno and Micro.

Whatever the code is, the inclusion of this line attachInterrupt(digitalPinToInterrupt(3), ISR_routine, FALLING); give the following error :

WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':
(.text+0x0): multiple definition of `__vector_7'
libraries/RBDDimmer/avr/RBDmcuAVR.cpp.o (symbol from plugin):(.text+0x0): first defined here

The vector number change if you use pin other than 3 on Arduino Micro.

siteswapjuggler, 2022-12-02:
Ok some investigation later : 

* commenting #define INT_vect lines in the header file make the compilation ok (I did not test the functionnal aspect as I debug this remotely for a friend without hardware to test it).

* seems like the interrupt method used in void dimmerLamp::ext_int_init(void) is not compatible with the Arduino standard which use attachInterrupt to keep track of the different IST method associated to each interrupt pins.

*  doing so the ISR(INT_vect) come in conflic with the core library

I will try some fixes but I will not have so much time to work on it correctly. This error could be common with other open issues so I hope it will help someone to correct it at one point :)

blackbird257, 2022-12-16:
Hi! , thank you for the research of the library issue. I have same issue with the interruptions (I use another device on port D3).
I'm trying to follow your advices, but I don't exactly know how to "declare ISR function before ext_int_init" - I just copied it before ext_int_init and renamed it to ISR_RBD. I get this error:
/mnt/create-efs/webide/16/d4/16d4ee6b75059aa8801ddf6d27b06fde:lubos257/libraries_v2/RBDdimmer/src/avr/RBDmcuAVR.cpp:80:8: error: expected constructor, destructor, or type conversion before '(' token

ISR_RBD(INT_vect)

^

/mnt/create-efs/webide/16/d4/16d4ee6b75059aa8801ddf6d27b06fde:lubos257/libraries_v2/RBDdimmer/src/avr/RBDmcuAVR.cpp: In member function 'void dimmerLamp::ext_int_init()':

/mnt/create-efs/webide/16/d4/16d4ee6b75059aa8801ddf6d27b06fde:lubos257/libraries_v2/RBDdimmer/src/avr/RBDmcuAVR.cpp:92:23: error: 'ISR_RBD' was not declared in this scope

attachInterrupt(INTx,ISR_RBD,RISING);

Could you be more specific with "declare ISR function before ext_int_init"  , 
2. question, down in the code of ccp file is also anothre ISR function: ISR (TIMER_COMPA_VECTOR(DIMMER_TIMER))
, should I rename also that one? 
Sorry for stupid questions, I'm beginner with programming.

Taprik, 2024-09-25:
I have conflict with the RFM69 library with the same rrors
`WInterrupts.c.o (symbol from plugin): In function `attachInterrupt':
(.text+0x0): multiple definition of `__vector_7'`
I try your solution.
It compiles but does not work.
Still investigating ...

siteswapjuggler, 2024-09-26:
Sorry it didn't work for you I was consulting for a client at the time, and do not have access to the project anymore to help you out. By the way my issue lack of descritption as I didn't even mention sur mcu and ide I was using... I hope you will find a solution.

Taprik, 2024-09-26:
Thank you siteswapjuggler,
I change 
 attachInterrupt(INTx,ISR_RBD,RISING);
to
attachInterrupt(digitalPinToInterrupt(7),ISR_RBD,RISING);
and it works
the INTx defined in the header was not the correct one for my leonardo

siteswapjuggler, 2024-09-26:
Well done thanks for sharing this solution :)