RobotDynOfficial/RBDDimmer #71: Issue when using more than one Dimmer on ESP32

frankiezafe, 2023-07-16:
Due to the initialiation of timer at each "begin", it's impossible to use more than one dimmer at the time with an ESP32.
I fixed the code by simply checking a static bool when begin in called.

It should be an easy fix, just copy/paste this cpp into `src/esp32/` folder, recompile and done.

[RBDmcuESP32.cpp.zip](https://github.com/RobotDynOfficial/RBDDimmer/files/12064744/RBDmcuESP32.cpp.zip)

All the best, thanks for the plugin, it's great!

cesincem, 2023-09-06:
I am also working on multiple dimmers. But I couldn't find a solution. I would appreciate it if you could share a sample 2 channel code with me.

hotswapster, 2024-06-02:
@frankiezafe can the same changes be made to ESP8266 code do you think?