RobotDynOfficial/RBDDimmer #72: Does't work with ESP8266

pritam2016bakshi, 2023-07-24:
stack>>>

ctx: cont
sp: 3fffff70 end: 3fffffd0 offset: 0010
3fffff80: 402012b2 feefeffe 0000000e 402027c5
3fffff90: 00002580 0000001c 00000000 4020133e
3fffffa0: 3fffdad0 00000000 3ffee7e0 40201058
3fffffb0: feefeffe feefeffe 3ffee834 40201e8c
3fffffc0: feefeffe feefeffe 3fffdab0 401010c5
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
�;⸮P⸮z�4z9⸮⸮⸮⸮⸮ISR not in IRAM!

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

Do have any solution??

Bsacho-za, 2023-08-13:
Im having the same issue

Bsacho-za, 2023-08-13:
found a fix

Edit the the file [src/esp8266/RBDmcuESP8266.cpp]

Add ICACHE_RAM_ATTR  in front of the function isr_ext(), 
in the same way as it is in the function onTimerISR() (so void ICACHE_RAM_ATTR isr_ext())

source 

https://github.com/RobotDynOfficial/RBDDimmer/issues/14

troyhowarth, 2024-03-01:
Thanks Bsacho-za! This did the trick for me.