RobotDynOfficial/RBDDimmer #77: i have i problem i need help

ZAKARIACL, 2024-06-07:
In file included from c:\Users\PC\Documents\Arduino\libraries\RBDdimmer\src/RBDdimmer.h:9,
                 from C:\Users\PC\Desktop\sketch_jun8a\sketch_jun8a.ino:1:
c:\Users\PC\Documents\Arduino\libraries\RBDdimmer\src/esp32/RBDmcuESP32.h:10:10: fatal error: esp_intr.h: No such file or directory
   10 | #include "esp_intr.h"
      |          ^~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

zetazetterberg, 2024-07-04:
replaced with esp_intr_alloc.h.

Arulamir, 2024-07-31:
> replaced with esp_intr_alloc.h.

after i change it the error going to file RBDmcuESP32.cpp
C:\Users\TOSHIBA\Documents\Arduino\libraries\RBDdimmer\src\esp32\RBDmcuESP32.cpp: In member function 'void dimmerLamp::timer_init()':
C:\Users\TOSHIBA\Documents\Arduino\libraries\RBDdimmer\src\esp32\RBDmcuESP32.cpp:53:27: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
   53 |         timer = timerBegin(0, 250, true);
      |                 ~~~~~~~~~~^~~~~~~~~~~~~~
In file included from C:\Users\TOSHIBA\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32/esp32-hal.h:84,
                 from C:\Users\TOSHIBA\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32/Arduino.h:36,
                 from C:\Users\TOSHIBA\Documents\Arduino\libraries\RBDdimmer\src\esp32\RBDmcuESP32.h:4,
                 from C:\Users\TOSHIBA\Documents\Arduino\libraries\RBDdimmer\src\esp32\RBDmcuESP32.cpp:3:
C:\Users\TOSHIBA\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
C:\Users\TOSHIBA\Documents\Arduino\libraries\RBDdimmer\src\esp32\RBDmcuESP32.cpp:55:29: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
   55 |         timerAttachInterrupt(timer, &onTimerISR, true);
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\TOSHIBA\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
C:\Users\TOSHIBA\Documents\Arduino\libraries\RBDdimmer\src\esp32\RBDmcuESP32.cpp:58:9: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
   58 |         timerAlarmWrite(timer, 30, true);
      |         ^~~~~~~~~~~~~~~
      |         timerWrite
C:\Users\TOSHIBA\Documents\Arduino\libraries\RBDdimmer\src\esp32\RBDmcuESP32.cpp:60:9: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
   60 |         timerAlarmEnable(timer);
      |         ^~~~~~~~~~~~~~~~
      |         timerAlarm

beubeu84, 2024-10-28:
I'm exactly in the same case

Benni22, 2024-12-01:
I have the same problem, can someone help?

JRSHENK, 2024-12-30:
I have had the same problem too. The only work around I have found so far was to roll the esp32 boards manager back from V3.x.x to any of the V2.x.x versions. Considering this library hasn't had any updates in about 5 years, I'm thinking they don't plan to fix it.