RobotDynOfficial/RBDDimmer #79: Request for compatability with new arduino uno R4.

brewystuff, 2024-07-12:
When trying to compile for arduino uno R4 this error is recieved. 

Arduino\libraries\RBDdimmer\src/RBDdimmer.h:21:3: error: #error "This library only supports boards with an AVR, ESP32, ESP8266, SAMD, SAM, STM32F1/F4 processor."
  #error "This library only supports boards with an AVR, ESP32, ESP8266, SAMD, SAM, STM32F1/F4 processor."
   ^~~~~
exit status 1

Compilation error: exit status 1

Grufoony, 2024-08-05:
I'd also like such implementation

matteogio, 2024-12-28:
I have the same probelm, it woluld be fantastic if the library colud support Arduino Uno R4 WiFi. It costs only 30€ and colud be the future of IOT!

computermouth, 2025-01-10:
Also came here looking for this

Wolper42, 2026-01-12:
I face the same problem and I would really appreciate a solution