RobotDynOfficial/RBDDimmer #81: ESP32 compliation error -  esp_intr.h

matteogio, 2025-01-10:
Trying to compile the example SerialMonitorDim for ESP32 DevModule on Arduino IDE 2.3.4 I obtain the compilation error "esp_intr.h - No such file or directory". 
The installed ESP32 board manager is 3.1.1, also with previous version I obtain the same error.

Anyone can help me? Thanks!

In file included from c:\Users\Documents\Arduino\libraries\RBDDimmer-master\src/RBDdimmer.h:9,
from C:\Users\AppData\Local\Temp\.arduinoIDE-unsaved2025010-6760-gx8a50.hjkw\SerialMonitorDim\SerialMonitorDim.ino:54:
c:\Users\Documents\Arduino\libraries\RBDDimmer-master\src/esp32/RBDmcuESP32.h:10:10: fatal error: esp_intr.h: No such file or directory
   10 | #include "esp_intr.h"
      |          ^~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

manumdk, 2025-01-25:
Hi, 
Are you find a solution ?

JRSHENK, 2025-01-27:
I don't have the best solution, but I do have a work around. In the Arduino IDF, under boards manager you must currently be using ver 3.x.x. If you roll it back to one of the 2.x.x versions it will work.

manumdk, 2025-01-28:
Hi,
Hi, I used this fork https://github.com/xlyric/RBDDimmer and managed to compile in version 3.2. 
Manu

Platform espressif32 @ 6.9.0 (required: espressif32 @ 6.9.0)
├── framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b (required: platformio/framework-arduinoespressif32 @ ~3.20017.0)
├── tool-esptoolpy @ 1.40501.0 (required: platformio/tool-esptoolpy @ ~1.40501.0)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5)

Libraries
├── RBDdimmer @ 0.0.0+sha.c3f1e99 (required: git+https://github.com/xlyric/RBDDimmer)

adamumarul, 2025-05-30:
> Trying to compile the example SerialMonitorDim for ESP32 DevModule on Arduino IDE 2.3.4 I obtain the compilation error "esp_intr.h - No such file or directory". The installed ESP32 board manager is [...]

i have a same problem and still have no solution

manumdk, 2025-05-30:
> i have a same problem and still have no solution

Yes, just above:  https://github.com/RobotDynOfficial/RBDDimmer/issues/81#issuecomment-2619964148